Protocentral/ProtoCentral_ads1262 #1: Specified folder/zip file does not contain a valid library

ecaits, 2016-12-28:
I have installed this library but when I compile the example code I got the error of Specified folder/zip file does not contain a valid library.

What can be the issue???

VenkateshBhat, 2016-12-28:
Please unzip the files, and place the src file in appropriate folder ,in the Arduino IDE, navigate to Sketch > Include Library.

ecaits, 2016-12-28:
Already unzip the files but no src file is found. Can you please suggest for that? 
Send me the link for that.

ecaits, 2016-12-28:
Now its working... There was a spelling mistake in example.ino file....
I have used MISO, MOSI, CS, SCK, Start and DRDY pin... I have not used PWDN pin as I have limited pins on boards....
Thanks for your support...

aravindaguru, 2018-02-16:
I have installed this library but when I compile the example code I got the error of Specified folder/zip file does not contain a valid library.i have unzip the files but no src file is found. Can you please suggest for that?

aravindaguru, 2018-02-16:
i am getting this error
 C:\Users\manoj12126\AppData\Local\Temp\Temp1_ProtoCentral_ads1262-master.zip\ProtoCentral_ads1262-master\Libraries\example\example.ino:24:28: fatal error: SoftwareSerial.h: No such file or directory  #include <SoftwareSerial.h>