Protocentral/ProtoCentral_ads1262 #4: Single-Ended mode

Sumsa, 2017-01-12:
Hi There,

Do you have any idea how to make this work in Single-Ended mode? i.e measure the voltage difference between input A0 and COM. The documentations says the 1262 should be able to operate in this mode, but I cant figure out how to switch when looking at the example. 

Along the same lines, where do I make changes if i want to measure from different pins than AN0 And AN1?

ecaits, 2017-01-13:
You need use INPMUX register to set the different analog pin as input.

Sumsa, 2017-01-17:
Thank you for pointing me in the right direction. Here is a small guide for people having the same problem. 

for single mode operation connect the negative differential pin to ground using a jumper wire. 

Change the pins used in line 78 in ads1262.cpp 

e.g change the hex value next to INPMUX to a binary number as stated on page 94 in the datasheet. eg 
ads1262_Reg_Write(INPMUX, B00000001); // will be AIN pin 0 and 1
ads1262_Reg_Write(INPMUX, B01110010); // will be AIN pin 7 and 2

To change the sampling rate to 10Hz where there is the highest db 50/60Hz noise reduction change mode 2 at line 62 to:
ads1262_Reg_Write(MODE2, B00000010);