Protocentral/ProtoCentral_ads1262 #5: ADS1262 activation issue

ecaits, 2017-01-31:
I am interfacing ADS1262 with Arduino mega2560 along with TFT, sd card and AD420.
The example given in library of ads1262 is working perfectly but when I interface ads1262 with other slave at that time, sometimes it activate and given output and sometimes it does not activate so not getting serial output.

Is there voltage related problem or anything else?

VenkateshBhat, 2017-02-01:
Hi ecaits,

Sorry for the inconvenience caused, could you please share your design? What other slave you are interfacing with?

ecaits, 2017-02-06:
I am interfacing TFT 2.2 SPI and SD Card along with ProtoCentral_ads1262 ADS1262.
When I run the code, the ADS1262 gives wrong value. Although it is giving perfect value when running independently.
What may be the issue. I have attached photo of interfacing design.
[Circuit.pdf](https://github.com/Protocentral/ProtoCentral_ads1262/files/754511/Circuit.pdf)