Protocentral/ProtoCentral_ads1262 #14: ADS1262 - Unable to acquire data

NikGreek, 2021-11-05:
Hello, I'm trying to connect the ads1262 with an arduino uno so I can measure voltage with higher precision.
I'm using the same connections you suggest and the same code.
However in serial monitor, I only get the "successful initialization" message and then nothing or I get only zero values of voltage.
I apply the analog signal from a NI ELVIS III breadboard.
Is this something you can help me with ?
Thank you in advance!

LAWBear, 2021-12-17:
I remain hopeful for suggestions/guidance on ADS1262 data values from ProtoCentral ADS1262 "1-Simple_Differential.ino".  Have upgraded MCU from Feather M0 to Feather M4 Express, and **commented** sketch "#include <SoftwareSerial.h> since previous comment.

Data "readout voltage/mv" remains 0.00 & 0.00 resp.

I embed logic analyzer image but cannot include *.rar archive of Saleae logic files for analysis.  Data only displays intermittently or at interval(s) of ~732ms??

[image]

Analyzer traces self labeled and arranged for clarity of response and issue.

PLEASE advise steps for further testing or diagnositcs.

LAWBear, 2022-02-07:
> Hello, I'm trying to connect the ads1262 with an arduino uno so I can measure voltage with higher precision. I'm using the same connections you suggest and the same code. However in serial monitor, [...]

LAWBear, 2022-02-07:
I MAY be getting close to successful configuration of the ADS1262 and the Feather M4 Express as an SD card data acquisition system.  This configuration also provides GPS header statements in each file, and fairly large SD cards (32Gb & 64Gb) using Fat32 or ExFat with 32k clusters and 128k clusters resp.  I am storing raw binary at up to 38.4 ksps to the SD card with some reasonable success, but still struggling a bit to get the ADS1262 to "start" and/or "sync" with the sample clocks.  To the interests of wizards I include my crude sketch.  Further, to some other inquiries, I have also successfully "tested" the ProtoCentral on Arduino Uno R3. and the "example" sketch.  (currently using pushbutton(s) to start file for selected DURATION)

Comments/Guidance appreciated:

``/*Compiled for Feather M4 Express (SAMD51)
Sample rate and ADS1262 tuning done in modified ads1262.cpp
Using data buffer sized for 32Gb Sandisk Extreme SD card
Sketch includes some place-holders for AirLift ESP32 Coprocessor (future work)*/

#include <SPI.h>
#include <SdFat.h>
#include <stdint.h>
#include <Adafruit_GPS.h>
#include <ads1262.h>
//#include <WiFiNINA.h>
#include <stdio.h>

#define SPI_SPEED SD_SCK_MHZ(12)
#define DURATION 10               // RECORDING DURATION
int count1=0;
SdFat SD;
File myFile;
char filename[15];

// name of the GPS hardware serial port?
#define GPSSerial Serial1

// Connect to the GPS on the hardware port
Adafruit_GPS GPS(&GPSSerial);

String NMEA1;
String NMEA2;
char c;

//AirLIft WiFi
const int WIFININA_SLAVESELECT = 13;
const int WIFININA_SLAVERESET = 12;
const int WIFININA_SLAVEREADY = 11;

const int ledStart = A1;
const int ledStop = A2;
const int btnStart = A3;
const int btnStop = A4;
unsigned long startRecTime = 0;
const int sd_cs_pin = 10;  //Feather M4 SD card chip-select pin
//const int GPO = ??       //not available until pushbuttons & LED's released for external controls

//Pins used for the connection with the sensor, the others are controlled by the SPI library):  (Feather M4 pins assigned in ads1262.h)
//const int ADS1262_DRDY_PIN = 6;
//const int ADS1262_CS_PIN = 9;
//const int ADS1262_START_PIN = 5;
//const int ADS1262_PWDN_PIN = A0;

int recPressed = 0;
int stopPressed = 0;
int flag = 0;

ads1262 PC_ADS1262;  //ADS class

//ADS1262 global
#define PGA 1                     // Programmable Gain = 1
#define VREF 2.50                 // Internal reference of 2.048Vc
#define VFSR VREF/PGA             
#define FSR (((long int)1<<23)-1)  

#define START1 0x08
#define STOP1 0x0A
#define RDATA1 0x12

//float volt_V=0;
[... 196 lines cut ...]
  while(!GPS.newNMEAreceived()) { //Keep reading characters in this loop until a good NMEA sentence is received
  c=GPS.read(); //read a character from the GPS
  }
GPS.parse(GPS.lastNMEA());  //Once good NMEA, parse it
NMEA1=GPS.lastNMEA();      //Once parsed, save NMEA sentence into NMEA1

while(!GPS.newNMEAreceived()) {  //Go out and get the second NMEA sentence, should be different type than the first one read above.
  c=GPS.read();
  }
GPS.parse(GPS.lastNMEA());
NMEA2=GPS.lastNMEA();
}  

void clearGPS() {  //Since between GPS reads, we still have data streaming in, we need to clear the old data by reading a few sentences, and discarding these
while(!GPS.newNMEAreceived()) {
  c=GPS.read();
  }
GPS.parse(GPS.lastNMEA());
while(!GPS.newNMEAreceived()) {
  c=GPS.read();
  }
GPS.parse(GPS.lastNMEA());
}
''
.